stm32duino/STM32LoRaWAN #5: unknown type name 'SUBGHZ_HandleTypeDef'

Lizethgm, 2022-12-13:
Hi, 
I was trying to compile on version 2.3 of the Generic STM32WL series board "Generic WLE5CCUx" and I got this error:

STM32LoRaWAN.cpp:1113:3: error: 'vcore_debug' was not declared in this scope; did you mean 'core_debug'?

I fix the error changing "vcore_debug" to "core_debug". I created a fork with this fixed #4  . But now when that was fixed appears new errors:

STM32LoRaWAN-main\src\BSP\subghz.h:50:8: error: unknown type name 'SUBGHZ_HandleTypeDef'
   50 | extern SUBGHZ_HandleTypeDef hsubghz;

STM32LoRaWAN-main\src\BSP\subghz.c:43:1: error: unknown type name 'SUBGHZ_HandleTypeDef'; did you mean 'UART_HandleTypeDef'?
   43 | SUBGHZ_HandleTypeDef hsubghz;

matthijskooijman, 2022-12-14:
Hi! I suspect that there might be some definitions missing in https://github.com/stm32duino/Arduino_Core_STM32 for your board. So far, we've only been testing with the Nucleo-WL55JC1 board.

Can you tell me a bit more about your board? Is it an off-the-shelf board (got a link?) or some custom board?

closed 2022-12-14 by fpistm

fpistm, 2022-12-22:
STM32 core v2.40 is released.