stm32duino/STM32LoRaWAN #7: RAK3172 "WLE5CCU6"

Lizethgm, 2022-12-21:
Hi,
I managed to run your library with rak3172 
For this I modified these parts:

In STM32LoRaWAN-main\src\BSP\radio_board_if.c
// Is a TCXO present on the board?
#if !defined(LORAWAN_BOARD_HAS_TCXO)
  #define LORAWAN_BOARD_HAS_TCXO 0U
#endif

  #if !defined(LORAWAN_RFSWITCH_PINS)
  #define LORAWAN_RFSWITCH_PINS PB8,PC13
  #define LORAWAN_RFSWITCH_PIN_COUNT 2
  #define LORAWAN_RFSWITCH_OFF_VALUES LOW,LOW
  #define LORAWAN_RFSWITCH_RX_VALUES HIGH,LOW
  #define LORAWAN_RFSWITCH_RFO_LP_VALUES LOW,HIGH
  #define LORAWAN_RFSWITCH_RFO_HP_VALUES LOW,HIGH

In STM32LoRaWAN-main\src\BSP\rtc.c
 hrtc.Instance = RTC;
  hrtc.Init.AsynchPrediv = 127;
  hrtc.Init.SynchPrediv = 255;
  hrtc.Init.OutPut = RTC_OUTPUT_DISABLE;
  hrtc.Init.OutPutRemap = RTC_OUTPUT_REMAP_NONE;
  hrtc.Init.OutPutPolarity = RTC_OUTPUT_POLARITY_HIGH;
  hrtc.Init.OutPutType = RTC_OUTPUT_TYPE_OPENDRAIN;
  hrtc.Init.OutPutPullUp = RTC_OUTPUT_PULLUP_NONE;
  hrtc.Init.BinMode = RTC_BINARY_ONLY;

How could I add this to your library without having to modify it, which is what I had to do?

 Also I tested your library as you had it in the beginning for the NUCLEO-WLE55JC1 board and everything worked correctly, I also tested it with the US915 and it still worked fine.

fpistm, 2022-12-22:
Hi,

I guess the simpliest way would be in the src\BSP\radio_board_if.c to define config between board definition:

```
#if !defined(LORAWAN_BOARD_HAS_TCXO)
#if defined(NUCLEO_WL55JC1)
  #define LORAWAN_BOARD_HAS_TCXO 1U
#elif defined(GENERIC_WL55CCUX) /* <-- or defined the RAK3172 variant in the core? */
  #define LORAWAN_BOARD_HAS_TCXO 0U
#endif
#endif
```
For RTC, default config is not correct?

Lizethgm, 2022-12-23:
Oh, thanks, I will try to do that

No, the rtc of the RAK3172 has two distinct things which are:
 hrtc.Init.AsynchPrediv = 127;
 hrtc.Init.SynchPrediv = 255;

Lizethgm, 2023-01-06:
Hi
Yes it worked, I put it this way to make it work:
In STM32LoRaWAN-main\src\BSP\radio_board_if.c

#if !defined(LORAWAN_BOARD_HAS_TCXO)
 #if defined(ARDUINO_NUCLEO_WL55JC1)  
  #define LORAWAN_BOARD_HAS_TCXO 1U

#elif defined(ARDUINO_GENERIC_WLE5CCUX) 
  #define LORAWAN_BOARD_HAS_TCXO 0U
#endif
#endif

#if !defined(LORAWAN_RFSWITCH_PINS)
  #if defined(ARDUINO_NUCLEO_WL55JC1)
  #define LORAWAN_RFSWITCH_PINS PC3,PC4,PC5
  #define LORAWAN_RFSWITCH_PIN_COUNT 3
  #define LORAWAN_RFSWITCH_OFF_VALUES LOW,LOW,LOW
  #define LORAWAN_RFSWITCH_RX_VALUES HIGH,HIGH,LOW
  #define LORAWAN_RFSWITCH_RFO_LP_VALUES HIGH,HIGH,HIGH
  #define LORAWAN_RFSWITCH_RFO_HP_VALUES HIGH,LOW,HIGH

#elif defined(ARDUINO_GENERIC_WLE5CCUX) 
  #define LORAWAN_RFSWITCH_PINS PB8,PC13
  #define LORAWAN_RFSWITCH_PIN_COUNT 2
  #define LORAWAN_RFSWITCH_OFF_VALUES LOW,LOW
  #define LORAWAN_RFSWITCH_RX_VALUES HIGH,LOW
  #define LORAWAN_RFSWITCH_RFO_LP_VALUES LOW,HIGH
  #define LORAWAN_RFSWITCH_RFO_HP_VALUES LOW,HIGH
#endif
#endif

In STM32LoRaWAN-main\src\BSP\rtc.c

#if defined(ARDUINO_NUCLEO_WL55JC1)
  hrtc.Instance = RTC;
  hrtc.Init.AsynchPrediv = RTC_PREDIV_A;
  hrtc.Init.OutPut = RTC_OUTPUT_DISABLE;
  hrtc.Init.OutPutRemap = RTC_OUTPUT_REMAP_NONE;
  hrtc.Init.OutPutPolarity = RTC_OUTPUT_POLARITY_HIGH;
  hrtc.Init.OutPutType = RTC_OUTPUT_TYPE_OPENDRAIN;
  hrtc.Init.OutPutPullUp = RTC_OUTPUT_PULLUP_NONE;
  hrtc.Init.BinMode = RTC_BINARY_ONLY;

#elif defined(ARDUINO_GENERIC_WLE5CCUX) 
  hrtc.Instance = RTC;
  hrtc.Init.AsynchPrediv = 127;
  hrtc.Init.SynchPrediv = 255;
  hrtc.Init.OutPut = RTC_OUTPUT_DISABLE;
  hrtc.Init.OutPutRemap = RTC_OUTPUT_REMAP_NONE;
  hrtc.Init.OutPutPolarity = RTC_OUTPUT_POLARITY_HIGH;
  hrtc.Init.OutPutType = RTC_OUTPUT_TYPE_OPENDRAIN;
  hrtc.Init.OutPutPullUp = RTC_OUTPUT_PULLUP_NONE;
  hrtc.Init.BinMode = RTC_BINARY_ONLY;
  #endif

tcpipchip, 2023-02-18:
@Lizethgm 
I am tryng to execute the AT/Arduino of RAK (RUI) on STM32WL55JC, but i am getting always JOIN TXD TIMEOUT...looks that the problem is some wrong RF SWITCH GPIO selected by me...
Can you give a tip ?
@fpistm 
Congratulations! I will test in our South Korean LoRaWAN module :)

tcpipchip, 2023-02-21:
I did now a blog using your STM32 LoRaWAN
https://lom204-cli-wisol.blogspot.com/2023/02/o-objetivo-deste-blog-e-demonstrar-como.html

tcpipchip, 2023-03-23:
And implement low power mode

fpistm, 2023-03-23:
> 

Part of another issue #9

fpistm, 2023-03-24:
@Lizethgm , @tcpipchip 
which RAK3172 you got?

fpistm, 2023-03-24:
Note that RAK provides Arduino support:
https://github.com/RAKWireless/RAKwireless-Arduino-BSP-Index

tcpipchip, 2023-03-24:
I dont have the RAK, i have the WISOL module...
I ported the RAK Arduino Code to run on WISOL.
But...it´s compatible,

ogimenez-smtc, 2023-03-24:
> Note that RAK provides Arduino support: https://github.com/RAKWireless/RAKwireless-Arduino-BSP-Index

Actually the RAK3172 Arduino support provided by RAK is only for RUI3 (RAK UI), which mean using a closed source library blob from RAK, it is not "low level" like what stm32duino is doing.

As per my understanding:

* RAK3172 Evaluation Board is a breakout board of the module
* RAK3172 Module is based on the STM32WLE5CC, only the RFO_HP is wired to the antenna
* RAK3172-SIP is based on the STM32WLE5JC, only the RFO_HP is wired to the antenna
* RAK3172LP-SIP is based on the STM32WLE5JC, only the RFO_LP is wired to the antenna

So what is your advice on the integration @fpistm ? Add custom code under `ARDUINO_RAK3172_MODULE`, `ARDUINO_RAK3172_SIP` and `ARDUINO_RAK3172LP_SIP` ?
Or add common definition under `ARDUINO_RAK3172` and only the RF switch config under dedicated defines?

Thanks

Oliv4945, 2023-03-24:
> Note that RAK provides Arduino support: https://github.com/RAKWireless/RAKwireless-Arduino-BSP-Index

Actually the RAK3172 Arduino support provided by RAK is only for RUI3 (RAK UI), which mean using a closed source library blob from RAK, it is not "low level" like what stm32duino is doing.

As per my understanding:

* RAK3172 Evaluation Board is a breakout board of the module
* RAK3172 Module is based on the STM32WLE5CC, only the RFO_HP is wired to the antenna
* RAK3172-SIP is based on the STM32WLE5JC, only the RFO_HP is wired to the antenna
* RAK3172LP-SIP is based on the STM32WLE5JC, only the RFO_LP is wired to the antenna

So what is your advice on the integration @fpistm ? Add custom code under `ARDUINO_RAK3172_MODULE`, `ARDUINO_RAK3172_SIP` and `ARDUINO_RAK3172LP_SIP` ?
Or add common definition under `ARDUINO_RAK3172` and only the RF switch config under dedicated defines?

Thanks

fpistm, 2023-03-25:
I guess adding `ARDUINO_RAK3172 `is the simpliest way and define correct RF definition for each sub-variant.

tcpipchip, 2023-03-25:
add too the LSM110A :)