stm32duino/STM32LoRaWAN #8: Generic Node Support

HelgeSeidel, 2023-01-04:
I’m extremely happy about the release of the STM32LoRaWAN library. By default the  STM32LoRaWAN library supports the STM32 Nucleo board with STM32WL55JC MCU.

I would appreciate if the STM32LoRaWAN library could support the Generic Node (https://www.genericnode.com/, https://github.com/TheThingsIndustries/generic-node-se) as well.

The pinout differences are:
RF_CTRL1 NUCLEO - PC4 —> Generic Node - PA0
RF_CTRL2 NUCLEO - PC5 —> Generic Node - PA1
RF_CTRL3 NUCLEO - PC3 —> Generic Node - PB8

Would be great if you could add the Generic Node support.

fpistm, 2023-01-05:
Hi @HelgeSeidel 
Yes it should be fine.
I know some tests have already been done by @sabas1080
https://github.com/TheThingsIndustries/generic-node-se/issues/206#issuecomment-1361891245

It is a community library, do not hesitate to contribute.
I do not have this board so hard to test.

Lizethgm, 2023-01-06:
Hi,
That's right, we were testing with the Generic Node and it worked by placing directly the generic node chip which is the variant "WL55CCUx". If we select directly the variant "Generic Node SE (TTI)" board it does not work, it marks error when starting the rtc.
Error: .......\Arduino\libraries\STM32LoRaWAN-main\src\BSP\rtc.c

The error mentioned is:
if (HAL_RTC_Init(&hrtc) != HAL_OK)
  {
    Error_Handler();
  }

What was changed was:
In STM32LoRaWAN-main\src\BSP\radio_board_if.c

#if defined(ARDUINO_GENERIC_NODE_SE_TTI) || defined(ARDUINO_GENERIC_WL55CCUX)
  #define LORAWAN_RFSWITCH_PINS PB8,PA0,PA1
  #define LORAWAN_RFSWITCH_PIN_COUNT 3
  #define LORAWAN_RFSWITCH_OFF_VALUES LOW,LOW,LOW
  #define LORAWAN_RFSWITCH_RX_VALUES HIGH,HIGH,LOW
  #define LORAWAN_RFSWITCH_RFO_LP_LP_VALUES HIGH,HIGH,HIGH
  #define LORAWAN_RFSWITCH_RFO_HP_HP_VALUES HIGH,LOW,HIGH

fpistm, 2023-03-23:
Hi @Lizethgm and @HelgeSeidel 
I've added the RFSWITCH pins definitions to the Generic node SE variant:
https://github.com/stm32duino/Arduino_Core_STM32/commit/5a3f412d385c386eb421ede38c4223dbab5d83df

fpistm, 2023-03-23:
I've been able to test and no RTC init error.
STM32 core version 2.5.0 will be released in the coming days including the RF SWITCH definition.
So I close this issue has thanks the core update the library is compatible with GNSE.

fpistm, 2023-04-11:
> I've been able to test and no RTC init error. STM32 core version 2.5.0 will be released in the coming days including the RF SWITCH definition. So I close this issue has thanks the core update the library is compatible with GNSE.

Hi @HelgeSeidel 
I've made some new test and get the rtc init error. After some investigation I found it is due to clock config issue, LSE was not initialized that's why we got this error. I've made a fix to the core: https://github.com/stm32duino/Arduino_Core_STM32/pull/1999