stm32duino/STM32LoRaWAN #9: LoRaWAN with low power and RTC

Lizethgm, 2023-01-16:
Hi, 
I was trying to use your example of sending data via lorawan and the library "STM32LowPower.h" and I get an error that two functions have multiple definitions. 

```
.: ../../../..\libraries\STM32duino_RTC\rtc.c.o: in function `HAL_RTC_AlarmAEventCallback':
rtc.c:(.text.HAL_RTC_AlarmAEventCallback+0x0): multiple definition of `HAL_RTC_AlarmAEventCallback'; ..\..\..\.\libraries\STM32duino_RTC\rtc.c.o: in function `RTC_Alarm_IRQHandler':
rtc.c:(.text.RTC_Alarm_IRQHandler+0x0): multiple definition of `RTC_Alarm_IRQHandler'; 
..\..\..\.\libraries\STM32LoRaWAN-main\BSP\rtc.c.o:rtc.c:(.text.RTC_Alarm_IRQHandler+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
```
Is crashing with another library that has the low power that is "STM32RTC.h" it is because they have the same functions.
How can I fix this crash?

fpistm, 2023-01-17:
Hi @Lizethgm 
This is a know restriction.
Currently, STM32LoRaWAN library implements itself its RTC usage.
As the STM32LowPower is dependent of the STM32RTC then you can't use it.
It is planned to use STM32RTC instead of own implementation for the STM32LoRaWan and the STM32LowPower.

jpmeijers, 2023-04-25:
I also stumbled upon this issue, getting the same error, after trying to follow the documentation on [Low Power Applications](https://stm32duino.github.io/STM32LoRaWAN/#autotoc_md8)

```
Linking .pio/build/STM32WLE5/firmware.elf
/home/jpmeijers/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/STM32WLE5/lib132/libSTM32duinoLoRaWAN.a(timer_if.c.o): in function `HAL_RTC_AlarmAEventCallback':
/home/jpmeijers/Documents/PlatformIO/Projects/LoraE5test/.pio/libdeps/STM32WLE5/STM32duinoLoRaWAN/src/BSP/timer_if.c:383: multiple definition of `HAL_RTC_AlarmAEventCallback'; .pio/build/STM32WLE5/liba52/STM32duino RTC/rtc.c.o:/home/jpmeijers/Documents/PlatformIO/Projects/LoraE5test/.pio/libdeps/STM32WLE5/STM32duino RTC/src/rtc.c:933: first defined here
/home/jpmeijers/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/STM32WLE5/lib132/libSTM32duinoLoRaWAN.a(rtc.c.o): in function `RTC_Alarm_IRQHandler':
/home/jpmeijers/Documents/PlatformIO/Projects/LoraE5test/.pio/libdeps/STM32WLE5/STM32duinoLoRaWAN/src/BSP/rtc.c:148: multiple definition of `RTC_Alarm_IRQHandler'; .pio/build/STM32WLE5/liba52/STM32duino RTC/rtc.c.o:/home/jpmeijers/Documents/PlatformIO/Projects/LoraE5test/.pio/libdeps/STM32WLE5/STM32duino RTC/src/rtc.c:961: first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/STM32WLE5/firmware.elf] Error 1
```

FRASTM, 2023-05-09:
The RTC of the stm32WL  will be configured in **MIX** mode (RTC_ICSR register BIN = [10])  
--> The calendar RTC_TR and RTC_DR are available 
--> the RTC SSR subsecond resgister is available on 32 bits clocked by ck_apre

In RTC Mix mode, the RTC_TR and RTC_DR calendar registers contents can be read/write in binary or bcd format (example : minutes expressed on 8 bits from 0-59 vs minutes expressed in one minutes-tens value plu one  minutes-units value) 
The RTC_TR and RTC_DR calendar registers format is _not_ the RTC Mix or BCD or Binary-only mode.

PR #25 (2023-10-12): Use the RTC in MIX mode and convert timeout values
When the RTC is using subsecond as a nb of milliseconds (cf PR https://github.com/stm32duino/STM32RTC/pull/95)

The LoraWan is always  using timeout values as a nb of ticks
Ticks is the RTC count unit when running in BINATY or MIX mode to be 1000ms/fqce_apre = 1000 / 256 ~ 3.9ms when the RTC is clocked by the LSE (32768Hz) 
fqce_apre = LSE clock/ (Async prediv + 1) 

Setting the Lorawan alarm (ALARM B) to expire in a nb of ticks, requires a conversion to when the RTC_StartAlarm() has a subsecond parameter expressed in milliseconds : 
Timeout i [... cut]
closed 2023-10-12 by fpistm