stm32duino/STM32LoRaWAN #14: Assync OTA

tcpipchip, 2023-02-21:
Hi, 

the basic.ino works cool, always JOIN to network...

But the simpleAsync.ino always fail to join.

Did you tested?

I am using on STM32WL55JC

fpistm, 2023-02-21:
Hi,
it was testes by @matthijskooijman and know to work.

tcpipchip, 2023-02-21:
wow
And with the most updated version ?

fpistm, 2023-02-21:
AFAIK yes.

tcpipchip, 2023-02-21:
:(

fpistm, 2023-03-23:
Small update, I'm waiting LoRa network access, I will test when available.

tcpipchip, 2023-03-23:
Basic still works (JOIN)
Assync not works (JOIN FAIL)

fpistm, 2023-03-23:
That is normal as no changes done on the lib. I've just warn you that I will test when I can have access to LoRa network. 😉

tcpipchip, 2023-03-23:
☺️

fpistm, 2023-04-11:
Hi @tcpipchip 
I've now access to LoRa network and tested the SimpleAsync example with Nucleo-WL55JC1 and it works as expected. I'm able to join and send packet while the LED is blinking.

tcpipchip, 2023-04-11:
thank to your feedback. 
Here still not work...never join :(

fpistm, 2023-04-11:
Well you should try to debug. Or at least enable the core log debug to see if you get some useful info.

tcpipchip, 2023-04-11:
how to enable the debug ?
I am using WISOL LSM110A. 
We are representative in South America.

fpistm, 2023-04-11:
To get the core logs, select in the menu debug and trace.

tcpipchip, 2023-04-11:
where is the menu debug trace ?
[image]

fpistm, 2023-04-11:
Have you a link to the board. 

Select "Core logs Enabled" in the "Debug symbols and core logs" menu where you have all options for the board...

fpistm, 2023-04-12:
> We are representative in South America.

Why would you mean by this? 

Seems this repo hold support for your board:
https://github.com/Support-SJI/LSM110A

So looking at the project above, you will need to create your own variant or at least configure properly RF Switch and RBI_CONF.
https://github.com/Support-SJI/LSM110A/blob/f20ba0fa4edeb9a54b144f254fadd0cfc4ac27b0/Projects/NUCLEO-WL55JC/Applications/LoRaWAN_SigFox/LSM1x0A/Core/Inc/radio_board_if.h#L62

So I close this issue as it is not linked to an issue with the library but wrong board config. Please next time be more precise on the hardware you used.

closed 2023-04-12 by fpistm

tcpipchip, 2023-05-26:
hi, i am back on tests...
I did the modifications of
https://github.com/Support-SJI/LSM110A/blob/f20ba0fa4edeb9a54b144f254fadd0cfc4ac27b0/Projects/NUCLEO-WL55JC/Applications/LoRaWAN_SigFox/LSM1x0A/Core/Src/radio_board_if.c
To be equal to the radio_board_if.c STM32LORAWAN
On BASIC  examplegetting this results
[image]
I observed that i am having RX_TIMEOUT, but the package is reaching the TTN
[image]
On SimpleAsync get JOIN FAILED in a short time...maybe is not tryng again before send this message ?
look
[image]

tcpipchip, 2023-05-26:
i will not cancel after fist try to JOIN with FAIL got...
Will try again in loop
I observed that always connects if try again...

tcpipchip, 2023-05-27:
solved the RX_TIMEOUT
It´s was wrong SUBBAND choose by me...
but...my opnion
Make the source code this way
while(1)
  {
    //modem.setDevAddr("00000000");
    //modem.setNwkSKey("00000000000000000000000000000000");
    //modem.setAppSKey("00000000000000000000000000000000");
    //modem.joinABPAsync();
    modem.joinOTAAAsync();  
    wait_for_idle();
    if (modem.connected()) {
      Serial.println("Joined");
      return;
    } else {
      Serial.println("Join failed");
      //while (true) /* infinite loop */;
      delay(10000);
      }
  }

tcpipchip, 2023-05-27:
Does the RX_TIMEOUT occur when the setup package  to be NOT CONFIRMED ?

tcpipchip, 2023-05-27:
[image]

tcpipchip, 2023-05-28:
backing to tests on LSM110A

tcpipchip, 2023-05-28:
Now all working
Look the blog
I had to modify the **radio_board_if.c**
and

```
void setup()
{
  pinMode(PA8, OUTPUT);
  Serial.begin(115200);
  Serial.println("Start");
  modem.begin(AU915);

  // Configure join method by (un)commenting the right method
  // calls, and fill in credentials in those method calls.
  modem.setAppEui("0000000000000099");
  modem.setAppKey("FF9F138B40180AA45D6846E0A0146954");
  modem.setDevEui("0080E115051FD80A");
  while(1)
  {
    //modem.setDevAddr("00000000");
    //modem.setNwkSKey("00000000000000000000000000000000");
    //modem.setAppSKey("00000000000000000000000000000000");
    //modem.joinABPAsync();
    modem.joinOTAAAsync();  
    wait_for_idle();
    if (modem.connected()) {
      Serial.println("Joined");
      return;
    } else {
      Serial.println("Join failed");
      //while (true) /* infinite loop */;
      delay(10000);
      }
  }

See the blog
https://lom204-cli-wisol.blogspot.com/2023/05/o-objetivo-deste-blog-e-demonstrar-como_26.html
}
```

tcpipchip, 2023-05-30:
Plans to offer p2p ?

fpistm, 2023-05-30:
> Plans to offer p2p ?

What do you mean by p2p?
Connect 2 boards thanks LoRa? If yes then you can use: https://github.com/jgromes/RadioLib
@matthijskooijman add the support of STM32WL with some examples: https://github.com/jgromes/RadioLib/pull/649

tcpipchip, 2023-05-30:
Thank you so much!!!!!!!!!!!!!!!!!!!!!!!

tcpipchip, 2023-07-18:
looks that is a problem here
 bool setAppEui(String value) { return setDevEui(value.c_str()); }

Should be

bool setAppEui(String value) { return setAppEui(value.c_str()); }

fpistm, 2023-07-18:
Hi @tcpipchip 
please submit an issue or a PR in order to track and fix this. 😉