stm32duino/STM32LoRaWAN #15: CW mode support?

Miceuz, 2023-03-09:
Is there a possibility to enter into continuous wave mode for testing purposes?

fpistm, 2023-03-13:
Hi @Miceuz

honestly, I don't know. Will try to get answer internally.

Miceuz, 2023-03-15:
@fpistm never mind, I have hacked it:

```c
bool STM32LoRaWAN::continuousWave() {
  MlmeReq_t mlmeReq;
  mlmeReq.Type = MLME_TXCW_1;
  mlmeReq.Req.TxCw.Frequency = 868100000;
  mlmeReq.Req.TxCw.Power = 15;
  mlmeReq.Req.TxCw.Timeout = 65535;
  LoRaMacStatus_t res = LoRaMacMlmeRequest(&mlmeReq);
  if (res != LORAMAC_STATUS_OK)
    return failure("Failed to enable CW mode: %s\r\n", toString(res));

  return true;
}
```
Would you be interested in adding it to `STM32LoRaWAN` class in a properly parametrized form? I could create you a PR.

matthijskooijman, 2023-03-15:
@Miceuz How does CW work after you enabled it? Does it just receive packets continuously? E.g. something like class C operation? Or is it a lower level of operation somehow?

Miceuz, 2023-03-15:
@matthijskooijman  No, it just outputs a continuous wave - necessary for antenna testing and for compliance testing in a lab. 

It's a special use case necessary when you are releasing a product, but not related to a normal use in any way.

matthijskooijman, 2023-03-15:
Ah, it's output, not input. I see.

Sounds like a useful and very well-contained addition, so a pull request would be welcome (with, as you mentioned, proper parameters and of course a bit of documentation in doxygen comments).

Miceuz, 2023-03-15:
PR #16

closed 2023-05-03 by fpistm