stm32duino/STM32LoRaWAN #17: No random number

Edzelf, 2023-03-22:
I use this library for a "Seeed Lora-E5 mini" board.  I had to make some changes for the 3 pins for the RF switches.  I also had to fix the  definition of "LORAWAN_TX_CONFIG", which should be  "RBI_COF_RFO_HP" for this board.

The module works well: the OTAA JOIN was successful and several packets were sent.

After a reset, a new JOIN was not possible.  The TTN error was “DevNonce has already been used”.

I found out that the devnonce was always "0000", so not very random.  The problem is the function "SUBGRF_GetRandom()" in radio_driver.c.  This function always returns a 32-bit zero value.

fpistm, 2023-03-23:
Hi @Edzelf 
seems already point on the GitHub STM32CubeWL repo: 
https://github.com/STMicroelectronics/STM32CubeWL/issues/46
unfortunately not reproduced by our team. 

I'm unable to test as I'm waiting an access to LoRa network but will have a look when available.

matthijskooijman, 2023-03-23:
I'm pretty sure that random numbers did work when I was working on this library (I remember it did not work because I messed up the code and it worked when I fixed the code - fix is in this repo already since the first publish).

Looking at the report linked by @fpistm this might be related to an incorrect timeout, but that probably only surfaces in specific timings when the radio is busy when requesting a random number, which could explain why this only happens in some cases and is not very reproducible.

@Edzelf, could you try changing the [value of the `SUBGHZ_DEFAULT_TIMEOUT` constant](https://github.com/stm32duino/Arduino_Core_STM32/blob/8d11df8d6136ce2b61636d5d3b9c2052719aaed5/system/Drivers/STM32WLxx_HAL_Driver/Src/stm32wlxx_hal_subghz.c#L151) to `1000U` and see if that helps? That file can be found in `~/.arduino15/packages/STMicroelectronics/hardware/stm32/2.4.0/system/Drivers/STM32WLxx_HAL_Driver/Src/stm32wlxx_hal_subghz.c` (on Linux, on Windows [it is in a similar place in your AppData](https://support.arduino.cc/hc/en-us/articles/360018448279-Open-the-Arduino15-folder)).

This is not a proper fix, but at least helps confirm the timeout is indeed the issue.

Edzelf, 2023-03-24:
Changing the time-out does not make any difference.  The result is always zero.

fpistm, 2023-04-07:
FYI, I've added the [LoRa-E5 mini](https://wiki.seeedstudio.com/LoRa_E5_mini/) to the core: https://github.com/stm32duino/Arduino_Core_STM32/pull/1997

I've tested and had no issue to join after several reset.
I've enable the TCXO in the variant like this:
```
// LoRaWAN definitions
#define LORAWAN_BOARD_HAS_TCXO          1U
#define LORAWAN_BOARD_HAS_DCDC          1U
#define LORAWAN_TX_CONFIG               RBI_CONF_RFO_HP

#define LORAWAN_RFSWITCH_PINS           PA4,PA5
#define LORAWAN_RFSWITCH_PIN_COUNT      2
#define LORAWAN_RFSWITCH_OFF_VALUES     LOW,LOW
#define LORAWAN_RFSWITCH_RX_VALUES      HIGH,LOW
#define LORAWAN_RFSWITCH_RFO_LP_VALUES  HIGH,HIGH
#define LORAWAN_RFSWITCH_RFO_HP_VALUES  LOW,HIGH

So, I close this issue as it seems linked to the configuration.

```

closed 2023-04-07 by fpistm

Edzelf, 2023-04-07:
Thanks, I will test it.

Edzelf, 2023-04-11:
I tried to install the new version.
In my platformio.ini:
```
 platform = https://github.com/stm32duino/Arduino_Core_STM32
 board = LoRa-E5 mini
```
Error message:
`Error: Could not find one of 'platform.json' manifest files in the package`

fpistm, 2023-04-11:
Pio is not  updated.  The ne variant for the LoRa  E5 mini is available in the main branch of the stm32 core. You can use it with Arduino IDE. Wiki explained how to use git repo.