stm32duino/STM32LoRaWAN #30: rtc init

tcpipchip, 2023-07-22:
Hi team
Did you already got this problem ?
Error: C:\Users\Usuario\Documents\Arduino\libraries\STM32LoRaWAN-main\src\BSP\rt
c.c (56)

closed 2023-07-22 by fpistm

fpistm, 2023-07-24:
Hi @tcpipchip 
I close this issue as no enough info and if you've search in the issues you would find this:
https://github.com/stm32duino/STM32LoRaWAN/issues/8#issuecomment-1503366955

So check the clock config of the board you used (don't know which one) to see if LSE is well initialized. This kind of issue will be avoided when #9 will be implemented as RTC will be managed by STM32RTC and not hardly by the library.

tcpipchip, 2023-07-24:
no problem Frederic!
Something is making me crazy! 
If i burn the my APP and make the reset...all works fine! I can do many resets and always the RTC works! And LoRaWAN too! (ABP, OTA)
But, if you make  RESET using the  power down the processor and power up again...got error on line 56 of RTC.c and timer freeze on APP...

  if (HAL_RTC_Init(&hrtc) != HAL_OK)
  {
    Error_Handler(); <====
  }

This is my reference of config (LSM110A)

fpistm, 2023-07-24:
When you reset and power always present, the RTC and clock are always on (backup domain) which could explain your issue. RTC uses the LSE and it is  probably not ready when you plug on. I would advice to add the LSE config in the SystemCoreClock config to avoid this issue like the fix I've made for an other board: 
https://github.com/stm32duino/Arduino_Core_STM32/pull/1999/files#diff-5b2c788f945e1c35895603854f029fe27b1478d4c96782acecb08fe752f3b7dfR73-R76

tcpipchip, 2023-07-24:
hi Fraderic

Now STM32LoRaWAN is 100% working on LSM110A :)

`void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

  HAL_PWR_EnableBkUpAccess();

  __HAL_RCC_LSEDRIVE_CONFIG(RCC_LSEDRIVE_LOW);
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE2);

  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI|RCC_OSCILLATORTYPE_LSE;
  RCC_OscInitStruct.LSEState = RCC_LSE_ON;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;

  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_NONE;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }

  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK3|RCC_CLOCKTYPE_HCLK
                              |RCC_CLOCKTYPE_SYSCLK|RCC_CLOCKTYPE_PCLK1
                              |RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_HSI;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.AHBCLK3Divider = RCC_SYSCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_0) != HAL_OK)
  {
    Error_Handler();
  }
}`

tcpipchip, 2023-07-24:
Btw, low power snergy (SLEEP) is implemented ?

fpistm, 2023-07-24:
> Btw, low power snergy (SLEEP) is implemented ?

No. Stm32rtc library is required before manage low power