stm32duino/STM32LoRaWAN #31: LoRaWAN Issue on RAK3172

HelgeSeidel, 2023-09-08:
Trying to run the LoRaWAN Basic example on my RAK3172 with STM32LoRaWAN library v0.1.1. Similar to https://github.com/stm32duino/STM32LoRaWAN/issues/24 I get the Message:

Start
Error\Arduino\libraries\STM32LoRaWAN\src\BSP\rtc.c (56)

fpistm, 2023-09-08:
Probably issue with LSE of the RAK. You should try to debug to see  what happen. Soon RTC will be managed by the STM32RTC library.

HelgeSeidel, 2023-09-17:
I will spend some time debugging as soon as my new development board arrived.

fpistm, 2023-10-12:
Hi @HelgeSeidel 
Could you retry with the latest verison of the library, as now the RTC is managed by the STM32RTC https://github.com/stm32duino/STM32RTC/releases/tag/1.4.0

I close this one as you could not have the same error with the 0.2.0 version of the library.

closed 2023-10-12 by fpistm

HelgeSeidel, 2023-10-13:
Hi @fpistm , I used the latest release for my project today. 

👍🏼👍🏼👍🏼