stm32duino/STM32LoRaWAN #33: Missing header file error

paulrota, 2023-09-20:
When trying to compile any example of STM32duinoLoRaWAN library (basic, SendAndReceive, asyncs) on my RAK3172 module I get the error:

Arduino\libraries\STM32duinoLoRaWAN\src\BSP\mw_log.cpp:39:10: fatal error: core_debug.h: No such file or directory

I've been looking for that header file but it's not existing in my files. Am I missing any additional stm32 core library?

fpistm, 2023-09-20:
Hi @paulrota 
this file is part of the core.
Which core version you used and Arduino IDE?
Maybe the core was not properly installed you can try to reinstall it.

paulrota, 2023-09-20:
Hi @fpistm and thanks for the quick reply.
I don't know which core version I'm using (honestly I have no idea about it). The Arduino IDE version is 1.8.19.
Should I try with the latest Arduino IDE version? Also I would like to understand what the core version is. Is it included in the Arduino install package?

fpistm, 2023-09-20:
when you go to the board manager, which version is shown for the STM32 core?

paulrota, 2023-09-20:
stm32 core version is 2.6.0. I had it added using additional board manager URL:
https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json

paulrota, 2023-09-20:
Now I'm facing an error regarding the missing cmsis_compiler.h as missing