stm32duino/STM32LoRaWAN #34: RAK3172T TXCO configuration Issue

HelgeSeidel, 2023-10-15:
I’m happy that the STM32LoRaWAN library V0.2.0 was released a couple of days ago.

I tried the new library with my RAK3172 evaluation board and everything worked as expected. On the evaluation board a RAK3172E module is installed and I did a full chip erase before I flashed my own file. 

In parallel I have 25 nodes with a RAK3172T module installed. The RAK3172T module has an TXCO installed for a lower operation temperature and need a special configuration. I was not able to get the same program running on the RAK3172T module.

I used Board “LoRa Boards” and the Board Part Number “RAK3172 Module”.  STM32LoRaWAN library V0.2.0 and STM32RTC library 1.4 on Arduino 2.2.1.

I guess we need a new device for the RAK3172T module.

HelgeSeidel, 2023-10-15:
I changes the LoRaWAN definitions LORAWAN_BOARD_HAS_TCXO in [variant_RAK3172_MODULE.h](variants/STM32WLxx/WL54CCU_WL55CCU_WLE4C(8-B-C)U_WLE5C(8-B-C)U/variant_RAK3172_MODULE.h) from 0U to 1U and and everything is fine.

fpistm, 2023-10-16:
Hi @HelgeSeidel 
This is not an issue. I guess you have to define new board variant or a derivative from the RAK3172 module added by one contributor:
https://github.com/stm32duino/Arduino_Core_STM32/pull/2005