stm32duino/STM32LoRaWAN #35: LoRa send and receive without joining the Things Network

larswd, 2023-10-23:
Hi!

I am working on a LoRa based device using the Wio-E5-mini, and I am very interested in using this module for my sensor. Yet, I do find the joinOTAA using the things network a bit *too much* so to say for the desired functionality of my sensor. Hence, I was wondering if it was possible to just transmit signals via the antenna without going to the things network. I was thinking something along the line of 

```cpp
LoRaModem modem;

void setup(void)
{

    if (!modem.begin(EU868)){
        Serial.println(F("Failed to start module"));
        while(1){}
    }
    Serial.print(F("Version: "));
   Serial.print(modem.version());

    }
}

void loop(void)
{
    String msg= "This is a good message!";
    modem.beginPacket();
    modem.print(msg);
    int err = modem.endPacket(true);
}
```
This might be a dumb question so to say, but it was not immediately clear for me when reading the documentation. If I overlooked something, I would be grateful for a pointer in the right direction.  My question still remains, is this kind of communication supported with this module, and if so, could I get a pointer as to how this is performed? And if it is not supported, would it be possible to request this to be supported in a future edition?

fpistm, 2023-10-23:
Hi @larswd 
in fact you can use this library:
https://github.com/jgromes/RadioLib
The STM32Wl support was added in version https://github.com/jgromes/RadioLib/releases/tag/5.6.0

closed 2023-10-23 by fpistm

larswd, 2023-10-23:
Thanks a lot for the help!

fpistm, 2023-10-23:
Welcome

g6094199, 2025-07-25:
> Thanks a lot for the help!

also plz consider  teh difference of LoRa and LoRaWAN.

LoRa is the mac layer only. there are many simple trasmission exapples for it.
and LoRaWAN is the complete stack (eg chirpstack or the things network)