stm32duino/STM32LoRaWAN #42: la915

tcpipchip, 2024-07-07:
got success and ported to LA915. I rebuilt the AU915 files.

tcpipchip, 2024-07-15:
Still running well on LA915 (LATIN AMERICA)
sir, how can i access the LOG info
core_debug(
    "McpsIndication: ind=%s, status=%s, multicast=%u, port=%u, datarate=%u, pending=%u, size=%u, rxdata=%u, ack=%u, dncnt=%u, devaddr=%08x, rssi=%d, snr=%d, slot=%u\r\n",
    toString(i->McpsIndication), ### toString(i->Status), i->Multicast, i->Port,
    i->RxDatarate, i->IsUplinkTxPending, i->BufferSize, i->RxData,
    i->AckReceived, i->DownLinkCounter, i->DevAddress,
    status->Rssi, status->Snr, status->RxSlot);

Inside the Arduino code ?

fpistm, 2024-07-15:
Enable the debug.

tcpipchip, 2024-07-15:
Yes, but i want to get access to that Status from my main source code.

tcpipchip, 2024-07-15:
yes, of course
void STM32LoRaWAN::MacMcpsConfirm(McpsConfirm_t* c) {
  // Called after an Mcps request (data TX) when the stack becomes idle again (so after RX windows)
  strcpy(STAT, toString(c->Status));
  core_debug(
    "McpsConfirm: req=%s, status=%s, datarate=%u, power=%d, ack=%u, %s=%u, airtime=%u, upcnt=%u, channel=%u\r\n",
     toString(c->McpsRequest), toString(c->Status), c->Datarate, c->TxPower,
     c->AckReceived,
    #if (defined( LORAMAC_VERSION ) && ( LORAMAC_VERSION == 0x01000300 ))
      "retries", c->NbRetries,
    #elif (defined( LORAMAC_VERSION ) && ( LORAMAC_VERSION == 0x01000400 ))
      "trans", c->NbTrans,
    #endif /* LORAMAC_VERSION */
    (unsigned)c->TxTimeOnAir, (unsigned)c->UpLinkCounter, (unsigned)c->Channel);
  instance->last_tx_acked = c->AckReceived;
  instance->fcnt_up = c->UpLinkCounter;
}

On stm32lorawan.cpp

I want to compare if the toString(c->Status) has some value on Arduino code.

tcpipchip, 2024-07-15:
my workaround is create a extern variable...but i dont like this :)
extern char STAT[128];

void STM32LoRaWAN::MacMcpsConfirm(McpsConfirm_t* c) {
  // Called after an Mcps request (data TX) when the stack becomes idle again (so after RX windows)
  strcpy(STAT, toString(c->Status));

and in the C Arduino

extern char STAT[128]; to have access to that variable with Status of LoRaWAN mac

tcpipchip, 2024-07-15:
work around

`/**
 * This is a more complex asynchronous sketch that shows how the
 * asynchronous code could be used in a sketch that uses a scheduler of
 * some sort to decide what code to run when.
 *
 * This sketch defines its own very, very simple scheduler that is not
 * suggested to be used directly, but just serves to show how a sketch
 * that uses some scheduling mechanism can be used with the
 * asynchronous STM32LoRaWAN API and the MaintainNeeded callback.
 *
 * The behavior of the sketch is simple, it just blinks a LED
 * (asynchronously) and at the same time joins the network and transmit
 * a packet periodically.
 *
 * Revised BSD License - https://spdx.org/licenses/BSD-3-Clause.html
 */
#include <STM32LoRaWAN.h>
#include <IWatchdog.h>

STM32LoRaWAN modem;

char STAT[128];

/*********************************************************************
 * This part of the sketch defines a very simple scheduler and defines
 * two tasks that handle the actual work.
 *********************************************************************/
struct Task {
  unsigned long time; /* When to run, 0 == never */
  void (*callback)();
};

enum Tasks {
  BLINK_TASK,
  LORA_WORK_TASK,
  LORA_MAINTAIN_TASK,
  NUM_TASKS,
};

void do_blink();
void do_lora_maintain();
void do_lora_work();
void maintain_needed_callback();

Task tasks[NUM_TASKS] = {
  [BLINK_TASK] = {0, do_blink},
  [LORA_WORK_TASK] = {0, do_lora_work},
  [LORA_MAINTAIN_TASK] = {0, do_lora_maintain},
};

void run_scheduler()
{
  // Super-simple scheduler that just checks all tasks and runs
  // any that are due.
  for (size_t i = 0; i < NUM_TASKS; ++i) {
    if (tasks[i].time != 0 && (int)(millis() - tasks[i].time) >= 0) {
      tasks[i].time = 0;
      tasks[i].callback();
    }
  }
}

/*********************************************************************
 * This part of the sketch defines the blink task, which just toggles
 * a LED whenever it is called.
 * And this is the entry points of the sketch.
 *********************************************************************/
static const unsigned long BLINK_TASK_TIME = 1000; /* ms */
[... 139 lines cut ...]

void setup()
{
  Serial.begin(115200);
  Serial.println("Start");
  modem.begin(AU915);
  modem.setPort(1);
  modem.setADR(true);
  modem.power(14);
      //wdt
  IWatchdog.begin(32000000);
  modem.setMaintainNeededCallback(maintain_needed_callback);

  //pinMode(LED_BUILTIN, OUTPUT);

  //do_blink();
  do_lora_work();
}

void loop()
{
  run_scheduler();
     //WDT WAKE UP
  IWatchdog.reload();
}`

tcpipchip, 2024-07-15:
workaround

`void lora_done()
{
  // If, after calling maintain() the library is no longer
  // busy, then the asynchronous operation has completed,
  // so check its results.
  String STAT_R = String(STAT);
  //Serial.println("++++++++++++++++++++++++++++++++++++++++");
  //Serial.println(STAT_R);
  //Serial.println("++++++++++++++++++++++++++++++++++++++++");
  if (lora_state == TRANSMITTING) {
    if (STAT_R == "LORAMAC_EVENT_INFO_STATUS_OK")
       Serial.println("Sent packet");
    else
       Serial.println("Fail sent packet");    
    // Done transmitting
    process_rx();
    lora_state = IDLE;
    // Schedule transmission of next packet
    tasks[LORA_WORK_TASK].time = millis() + TX_INTERVAL;
  } else if (lora_state == JOINING) {
    if (modem.connected()) {
      Serial.println("Joined");
      send_packet();
    } else {
      Serial.println("Join failed");
      lora_state = IDLE;
      tasks[LORA_WORK_TASK].time = millis() + RETRY_JOIN_INTERVAL;
    }
  }
}`

fpistm, 2024-07-16:
Hi @tcpipchip 
Please try to format the code else it is really hard to read.

It is fine if you achieve to get the value you want.

fpistm, 2024-11-28:
Hi  @tcpipchip 
Any update on this ?

tcpipchip, 2024-11-28:
Hi, i config la915.c la915.h to run on stm32 cube ide. Works nice. Want you those files ?

fpistm, 2024-11-28:
If you think it is useful and you can test with this library. Feel free to submit a PR. 😉

tcpipchip, 2024-11-28:
I will first to  see if compiles…

fpistm, 2024-11-28:
Thanks, anyway you can share it here for ref.

tcpipchip, 2024-11-28:
I replaced the AU915 parameters by the LA915 parameters, both .C and .H
The true is that i modified the AU915.h and AU915.c files, i not created a new on the STM32LORAWAN ?
Why this ? Only to test...
If the user choose now AU915 on modem.begin(AU915), the true is that is using the LA915...
At moment i still not added a new REGION...then modem.begin(LA915) will not compile..
I hope to test tomorrow if works on LA915

tcpipchip, 2024-11-29:
I sucessfully access now the LA915 (LATIN AMERICA) Everynet with STM32LORAWAN (STM32WL55-LSM110A)
Now i will need your help Frederic
I can send the files AU915.C/AU915.H (both modifed to LA915) and you or team edit it to LA915.C and LA915.H
I dont have experience to do that...
[image]
[image]

fpistm, 2024-12-04:
Share your file. Then we will see if we could integrate it. :wink:

tcpipchip, 2024-12-04:
@fpistm 
The best news of the year :)
[STM32LoRaWAN-main.zip](https://github.com/user-attachments/files/18014659/STM32LoRaWAN-main.zip)
attched
And here the code 

`/**
 * This is a more complex asynchronous sketch that shows how the
 * asynchronous code could be used in a sketch that uses a scheduler of
 * some sort to decide what code to run when.
 *
 * This sketch defines its own very, very simple scheduler that is not
 * suggested to be used directly, but just serves to show how a sketch
 * that uses some scheduling mechanism can be used with the
 * asynchronous STM32LoRaWAN API and the MaintainNeeded callback.
 *
 * The behavior of the sketch is simple, it just blinks a LED
 * (asynchronously) and at the same time joins the network and transmit
 * a packet periodically.
 *
 * Revised BSD License - https://spdx.org/licenses/BSD-3-Clause.html
 */
#include <STM32LoRaWAN.h>
#include <IWatchdog.h>

STM32LoRaWAN modem;

char STAT[128];

/*********************************************************************
 * This part of the sketch defines a very simple scheduler and defines
 * two tasks that handle the actual work.
 *********************************************************************/
struct Task {
  unsigned long time; /* When to run, 0 == never */
  void (*callback)();
};

enum Tasks {
  BLINK_TASK,
  LORA_WORK_TASK,
  LORA_MAINTAIN_TASK,
  NUM_TASKS,
};

void do_blink();
void do_lora_maintain();
void do_lora_work();
void maintain_needed_callback();

Task tasks[NUM_TASKS] = {
  [BLINK_TASK] = {0, do_blink},
  [LORA_WORK_TASK] = {0, do_lora_work},
  [LORA_MAINTAIN_TASK] = {0, do_lora_maintain},
};

void run_scheduler()
{
  // Super-simple scheduler that just checks all tasks and runs
  // any that are due.
  for (size_t i = 0; i < NUM_TASKS; ++i) {
    if (tasks[i].time != 0 && (int)(millis() - tasks[i].time) >= 0) {
      tasks[i].time = 0;
      tasks[i].callback();
    }
  }
}

/*********************************************************************
 * This part of the sketch defines the blink task, which just toggles
[... 143 lines cut ...]

void setup()
{
  Serial.begin(115200);
  Serial.println("Start");
  modem.begin(AU915);
  modem.setPort(1);
  modem.setADR(true);
  modem.power(14);
      //wdt
  IWatchdog.begin(32000000);
  modem.setMaintainNeededCallback(maintain_needed_callback);

  //pinMode(LED_BUILTIN, OUTPUT);

  //do_blink();
  do_lora_work();
}

void loop()
{
  run_scheduler();
     //WDT WAKE UP
  IWatchdog.reload();
}`

fpistm, 2025-02-04:
Hi @tcpipchip 
Thanks for the greetings.

As I've worked on the library this day, I've dug into the requested region: LA915.
The LoRa Alliance does not specify an LA915 region. The region to be used in Brazil is AU915 specification.
As LA915 is not official from the Alliance, I will not add it or accept a PR for it. If ST add it to the STM32CubeWL LoRaWan implementation then it will be inherited here but don't think they will add it as not in the specifications.

You can simply apply your change in a fork.
So I close this issue.
Sorry.

closed 2025-02-04 by fpistm

tcpipchip, 2025-02-04:
hi @fpistm 

No problem...

Well, if someone in Brasil use the STM32LoRaWAN with LA915, contact me!

Thanks!