stm32duino/STM32LoRaWAN #49: Packet send failing on US915

naveen-79, 2025-01-15:
Sending LoRaWAN packet on `US915` frequency returns failed but sending the same on `EU868` returns success.  
I'm using the [lora basic](https://github.com/stm32duino/STM32LoRaWAN/blob/main/examples/Basic/Basic.ino) example code.  
Only the REGION has been altered to use US915.

fpistm, 2025-01-15:
Hi @naveen-79 
Unfortunately, I can't test with this band in France.
You best best is to try to debug but don't see why only the freq change the send behaviour.

naveen-79, 2025-01-15:
What are the steps i should follow to debug the issue

fpistm, 2025-01-15:
Enable the "Optimize for Debugging" in the "Sketch" menu then build and upload.
Finally start debug by clicking on "Start Debugging" icon.
Are you sure your network or gateway are compatible with US915 and properly configured?
Join is successful?

naveen-79, 2025-01-15:
The gateway is configured properly. I was able to send data using a different lora module.
Will try the debugging option.

fpistm, 2025-01-15:
Maybe link to the Duty cycle limitation:
https://stm32duino.github.io/STM32LoRaWAN/
Honestly, don't know very well this. Maybe @matthijskooijman have some input.

naveen-79, 2025-01-16:
found the issue. Wrong datarate setting.
[image]

naveen-79, 2025-01-16:
the frequencies on my gateway and node does not match. how do i fix the issue, should i change frequencies on the gateway or on the node. currently i'm using the aws iot core network server and all the frequencies on the gateway are pre populated.
[image]

fpistm, 2025-01-20:
Hi @naveen-79 
I'm not an expert of LoRa configuration. For data rate you can probaly try `dataRate()` api
https://stm32duino.github.io/STM32LoRaWAN/classSTM32LoRaWAN.html#a5cca097cfdebc2227eb4ff0cb5358084

About freq, you can check those value:

naveen-79, 2025-01-23:
changed the data rate `DR_3` got it working.

naveen-79, 2025-01-23:
i believe this issue can be closed

fpistm, 2025-01-23:
Thanks for the feedback.

closed 2025-01-23 by fpistm